fashberg/WThermostatBeca #113: BAC-002 and BECA-WIFI (BHT 1000)

jarek090, 2021-03-24:
Hello,

First of all I would like to thank you for providing great project.
I am currently building smart home system using PLC Siemens S7-300 and OpenHAB 3 and also BECA-WIFI thermostat.
I am using your v1.19.beta1-fas firmware and I am facing several issues:
- ESP disconnects form AP quite often (currently I have DHCP on router and I can not change it, it is ISP router) and its IP is changed,
- MQTT (I am using Mosquitto) messages come to server irregulary so sometimes temerature is not refreshed for long time,
- MQTT: updating targetTemperature (manually using mosquitto_pub) works only from time to time, it seems I have to "fire" command in proper time slot. I ws using Web diagnostic and I barely could see messages comming from MQTT server,
- Thermostat switches from  manual mode to schedule mode by itself, I want keep it in manual mode  
- Thermostat switches off itself from time to time 

I tested both BAC-002 and BECA-WIFI devices behave same way.
I would appreciate any guide or help to solve above issues. I also started to play with PlatformIO but probably it will take some time for me to go thru the code...

Notice: 
To program ESP for BAC-002 and BECA I had to unsloder resitor and transistor which I marked on attached picture:
[image]

Thank you !
Jarek

badger707, 2021-03-29:
For disconnects from AP (#1), it could be that your ISP router wifi signal is not strong enough at that spot and your thermostat keeps dropping off. Try to move thermostat closer to wifi AP and see if that helps. If this solves the dropout issue, think on getting wifi extender. Also, check [Settings] > [Configure Network] > "Enable Fallback to AP Mode..." -- keep unchecked.
As for #2-3 -- this could be related to #1 as mentioned above.
Manual/Auto switching (#4) -- check [Settings] > [Configure Thermostat] > "Switch back to auto..." -- keep unchecked.

Unsoldered components -- not sure you need to do that, just connect 4+1 wires straight from serial controller to ESP directly or via adapter to these pins, that worked for me all the time for multiple projects:
[image]
:

There is a good chance that you've fried these SMD transistors too....

mvdklip, 2021-03-30:
As set out in #115 I also had trouble programming the ESP while connected to the thermostat board. It seems some people are 
"lucky" to have a UART which is able to overshout the connected Sonix MCU because that one is trying to talk it's binary protocol to the ESP while you are trying to program it. I think the author of this issue disabled the MCU to ESP communication by desoldering those SMD components.

jarek090, 2021-04-15:
First of all thank you for all your comments.

@ badger707 - thank you for guides, unfortunatelly this does not solve my issues. I traced messages in web log and I can see that MQTT communication is irregular and connection is broken so I can not set temperature. MQTT functions does not try to reconnect after timout for some reason. It refreshes sometimes after 5 - 10 minutes.
MQTT server works fine all times. Thermostat is 1,5 meter from WiFi router. The other problem is that NTP does not refresh and after a day the time is about 30 minutes differance between true time (clock is to fast)

@mvdklip - I had to unsolder to brake the lilne between MCU and ESP. Most crucial in my case was resistor which you can see in the left corner of picture. It was not necessary to unsolder Q8 trasistor. I am using CH340 USB/UART converter.

closed 2022-10-16 by github-actions[bot]