fashberg/WThermostatBeca #117: Alternatives for german standard frames

coldstorm82, 2021-04-13:
Hello community. I run Home Assistant and do a lot of Tasmota stuff. Now I'm looking for wall mount thermostats on ESP basis to integrate them as this project describes.

But here is the problem: in my house the original thermostats are installed in those German standard double frames together with a light switch, which means the usual 86x86mm replacements just don't fit. I've google the internet, but couldn't find anything smaller. The only option is to integrate the homematic thermostats (maybe with raspberrymatic integration in HA). But they are way more expensive and look like 1998 😉.

Does anybody have an idea for a smaller alternative that fits in those frames or replaces a frame like 65x65mm (I mean, I don't want to open the wall...)?

emign, 2021-09-28:
> Hello community. I run Home Assistant and do a lot of Tasmota stuff. Now I'm looking for wall mount thermostats on ESP basis to integrate them as this project describes.
>
>
>
> But here is the problem: [...]

Have you found anything suitable ?

coldstorm82, 2021-09-28:
Unfortunately no... 
But honestly, after spending weeks of searching I stopped a couple of month ago. The post above was kind of my last try in the communities.

Still, I'd be very happy if anyone has a suggestion. 👍🏻

closed 2022-10-14 by github-actions[bot]