fashberg/WThermostatBeca #124: Is this a new version of bac-002-wifi?

kobi-wan, 2021-06-06:
Hello,
I ordered a set of new GALWW thermostats from Moes. After I opened the device, I saw a different  label than I suspected. It is labeled "ZWX-GA/GC-CONTROL" and it's from October 2020. Also the layout is a little bit different. The ESP seems the same (TYWE3S).
So my question is: does it work with ThermostatBecaWifi? Has anybody already flashed the firmware on these version? I don't know if I would try it, because I could still send it back.

[image]

tsrubar, 2021-06-06:
Yes, these are without backlight of central button, back side ribbon has different number of wires, but they flash just fine. I have done 7 of these 2 weeks ago. Imho enclosure looks and feels much cheaper, what is unfortunate.

kobi-wan, 2021-06-07:
Okay, thank you. I already noticed, that the central button has no light anymore. So it's better to get one of the older designs imho.

maierei, 2022-03-14:
Hello @tsrubar,

I also got the same device version but was not able flash the new firmware from Klaus Ahrenberg. I soldered the wires just like in his manual, but cant get any connection using a USB programmer. 
Do you have any hints for me maybe?
Which exact wiring did you use? 
Did you connect the  ESP-GPIO0 to GND of the programmer when powering the ESP?
Did you connect ESP-GPIO15 to GND of the programmer or is this not needed?

Which programmer did you use?

Thanks
Markus

tsrubar, 2022-03-14:
Hi,
it is kind of hard to remember right now, but iirc I connected 3.3V from programmer to Vcc and En on thermostat, Gnd from programmer to Gnd and GPIO0 on thermostat (to boot in flash mode) and Rx/TX on programmer to Tx/Rx on thermostat.
As a programmer I used cheap NodeMCU.

maierei, 2022-03-15:
@tsrubar 
Hi, thanks a lot - That's how I tried it - but I used a different programmer and also read that it coul be some kind of power drain - so I have ordered the same programming unit and will test it with that one.
Thanks for gettng back so quick and looking for your post.

maierei, 2022-03-16:
@tsrubar 
My new programmer arrived today, but it's sad it does not work.
I always get 
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
It looks like you have connected an additional BLUE cable somewhere on the board - what's that?

tsrubar, 2022-03-16:
> @tsrubar My new programmer arrived today, but it's sad it does not work. I always get A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header It looks like you have connected an additional BLUE cable somewhere on the board - what's that?

Oh, I am sorry to hear that. That blue cable is just GND from programmer connected to more accessible place on PCB of thermostat directly connected with GND on TYWE3S. I was trying hard to find the programmer, I used, but we've just moved in our new house and everything is just a mess packed in boxes.
If you have it connected as described, did you try to push reset button few times? Iirc, I had to do it from time to time, but in the end, i managed to flash all 9 of my thermostats. Could you take photo of your setup, so maybe I could possibly notice something that you could potentionally have missed?

maierei, 2022-03-16:
I just thought I got it working as I used 5V and hit GND with RST and GPIO0 a few times because suddenly I got a connection, but:
- Reading and erasing the flas was not supported
- Writing was supported and it took about 10 times until the counter run to 100% (the tests before stoped at 10 to 20%).

Now I have reassembled the device and hoped I get an Wifi from it to configere but it seems just nothing happened...

Any thougts?

tsrubar, 2022-03-16:
@maierei dont do it with 5V, you could burn it. Flashing procedure do erasing first and then writing. I was using tasmotizer to flash tasmota to it. You could possibly try to plug it in stronger usb port.
Just found it!
[image]

maierei, 2022-03-16:
@tsrubar - ok I will give it a try tomorrow when my girls look at "Germanys next Topmodel"...
So you used Tasmotizer and the firmware from here: https://github.com/klausahrenberg/WThermostatBeca ?
I will try Tasmotizer tomorrow.
Thanks
Markus

tsrubar, 2022-03-17:
@maierei I cant remember where I got if from, but as far as md5 check for downloaded file is ok, it doesnt really matter where will you download from :) What I exactlly used was: tasmotizer-1.2.exe, but its over 38mb in size, so i temp-hosted it here: https://www.file.io/Gtyi/download/0T8UfQa199gl

PatrickHuetter, 2022-08-08:
Did you get it to work? I have the same problem with newly arrived BAC-002 Wifi thermostats. I already tested 2 different programmers and i'm sure that i soldered everything correctly. Which programmer did you use?

kobi-wan, 2022-08-10:
@PatrickHuetter: I can't tell, because I exchanged it with the older one.

PatrickHuetter, 2022-08-10:
Thanks @kobi-wan . I ordered a [different programming device](https://amzn.to/3BTkHHo). Optically the same one that @tsrubar has used on his photos. Hopefully this will work since i have many thermostats of this model that i want to flash. I'm currently waiting for it to be delivered. I'll report wether it worked or not.

PatrickHuetter, 2022-08-11:
@maierei Did you get it to work? I have 3 different programmer devices now and none of them worked. I also opened another thermostat (bought many of them) and soldered again and tested with different computers. I can't get a connection to the ESP.

```
esptool.py -p /dev/cu.usbserial-10 -b 460800 read_flash 0x00000 0x100000 originalFirmware1M.bin                                                                                                       ✔
esptool.py v4.2
Serial port /dev/cu.usbserial-10
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

Does somebody have another idea what to do now? Maybe it's the same problem like described here:
https://github.com/fashberg/WThermostatBeca/issues/113#issuecomment-810168238
https://github.com/fashberg/WThermostatBeca/issues/113#issuecomment-820177953
https://github.com/fashberg/WThermostatBeca/issues/115#issuecomment-818619946

[image]

fashberg, 2022-10-02:
Hey!
Compared to 2019 Version some changes - see image below

Check my post here: https://github.com/fashberg/WThermostatBeca/issues/115#issuecomment-1264667961

The 5V MCU to 3,3V ESP conversion seems still to be there, so if the MCU is still at 5 Volt and not powered, because you only connect the 3.3volt lane then probably it's something else. Either not enough 3.3 Volt power or some kind of hardware flashing protection.

Where have you bought the new version?

[image]

PatrickHuetter, 2022-10-06:
I bought 7 pieces of them on aliexpress. I already tested 4 different programmers and today i tested with a 3d printed programming jig (the one from the flashing.md). Always the same problem. 

```
esptool.py --port /dev/cu.usbserial-A50285BI -b 460800 flash_id                                                                                         2 ✘  8s 
esptool.py v4.3
Serial port /dev/cu.usbserial-A50285BI
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

I dont't know what to do now. Maybe i have to desolder them or just buy the zigbee version and sell the wifi ones.

rlehfeld, 2022-11-06:
I brought 4 Moes BHT-002-GALW(WiFi) and the PCB looks like in [picture](https://user-images.githubusercontent.com/31758598/120930338-2a0d5e00-c6ed-11eb-8bdc-608e6abac4f0.jpg) from [tsrubar](https://github.com/tsrubar) apart from the fact that the MCU does not have any kind of labeling in none of the 4 devices. It also says ZWX-GA/GC-CONTROL 2020.10.17 First I also faced problems in flashing the device. I tried several things. What actually helped at the end was connecting the position in the middle of J2 with GND. This one is connected to one of the pins of the MCU. I believe it is either an enable or rst pin of the MCU. At least it worked for me. It might be helpfull for others as well.

So in total I had following cabling:

3V3 <----> VCC
3V3 <----> EN
GND <----> GND
GND <----> GPIO0
TX <----> RXD0 
RX <----> TXD0 
middle J2 <---> GND

Sub-7, 2022-12-22:
Successfully flashed four thermostats as in the first picture.
Wiring was as follows:
[image]

Kolia56, 2023-03-08:
Hello to everyone. I've been trying to flash an ME81H thermostat with exactly the same wiring as above. It is referenced at [bladadder](https://templates.blakadder.com/ME81H.html ) It features a WBR3 module instead of the TYWE3S. Pinout is the same. I'm using a CH340G as suggested by Tasmota. I've successfully flashed a Shelly 2.5 using this adapter. 
I tried many things before posting. I even tried to ground the RST pin of the [MCU](https://datasheet.lcsc.com/lcsc/1912111437_SOC-Shenzhen-SinOne-Microelectronics-SC92F8547P48R_C367200.pdf), still no success.
To start with I've been using epstool flash_id that returns connection error.
Does anyone have an idea how I could progress?

@PatrickHuetter: did you buy the programming jig at some place?

Thank you

Master-Rudi, 2023-03-08:
did you buy the programming jig at some place?

https://www.ebay.de/itm/134389256425?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=OzFj3xP8T7C&sssrc=2349624&ssuid=OzFj3xP8T7C&var=&widget_ver=artemis&media=COPY

PatrickHuetter, 2023-03-10:
> @PatrickHuetter: did you buy the programming jig at some place?
No, i printed it by myself with a 3d printer. Don't remember the 3d files link but it was mentioned here.

closed 2023-06-15 by github-actions[bot]