fashberg/WThermostatBeca #126: HA Autodiscovery doesn't work

michalpulda, 2021-06-21:
Hi, flashed the BHT-6000-GALW yesterday using a 3d-printed pogo-adapter and Tasmotizer. Worked very well.

I just couldn't make the HA Autodiscovery to work, though. However, adding the termostat manually to configuration.yaml works. 

I could not use the password for the default Mosquito user (homeassistant) because the firmware can't take a password this long. Thus I created a new user, put it to the Mosquitto configuration under the logins section and used it for the thermostat's config in the network section.

Anyone solved this? Thanks!

yasars, 2021-08-03:
You must set an password.. look into changelog from mqtt server
-Support for anonymous logins has been removed

Fran6u, 2021-12-27:
> 

Hi, mind advising me on how you added it "manually"? I've been trying without luck to get it into my HA. Many thanks

stav242, 2021-12-27:
Hi,
I tried a lot of things, including removing all mqtt config and starting
again... But a few steps worth trying..

1. Make sure your mqtt server is working
Check your ha mqtt server logs
And Use mqtt explorer to check.
I had the some other plugin which was conflicting withe with default mqtt
port... So changed the other app and restarted.

2. Check your mqtt devices are configured correctly.
I had a typo in the settings.
You should see activity in mqtt explorer.

3. Check ha mqtt integration is setup
Check the logs, make sure there are no errors
Make sure there is no mqtt config in th configuration file if you are using
the integration config.

Good luck.

closed 2022-10-09 by github-actions[bot]