fashberg/WThermostatBeca #127: BHT-002 does not reconnect to wi-fi router after router's reboot

paulius2k, 2021-09-15:
Hello,
thank you for your firmware!

The problem I have is that after reboot of wi-fi router, the BHT-002 does not reconnect to the wifi network.
The only way to fix it is to cut the power and power on the BHT-002.

I have the ```Enable Fallback to AP-Mode if WiFi Connection gets lost``` option **disabled**.

Can this issue be fixed? 
Or is there some workaround to resolve it?

Thank you!

bensuffolk, 2021-12-11:
I seem to have a similar problem, I have 2 of these thermostats one in good wife coverage the other in poor coverage. The one in poor coverage often drops out and only reconnects when I reboot the thermostat by cutting the power to it.

I also have the Enable Fallback  AP mode set to off, so I don’t think that’s the reason it’s not reconnecting.

it would be great if we could find away to debug what’s going on.

iMike78, 2022-03-03:
I experienced the same issue. Fortunately I don't need to restart my AP-s too often, but the problem still exists.

paulius2k, 2022-07-18:
This issue still persists.

paulius2k, 2022-10-17:
The issue is still there. Any chances anyone would look into this one? Thank you.

donivan2004, 2023-02-24:
Hello, you can try this solution https://github.com/fashberg/WThermostatBeca/issues/193 it works for me.

paulius2k, 2023-05-26:
Is there a compiled binary with #193 fix? I can't compile myself..

donivan2004, 2023-05-26:
> Is there a compiled binary with #193 fix? I can't compile myself..

I updated it to my Drive: https://drive.google.com/file/d/1uxBnz5NtyNye1kTVE_dXRqnkUNpsOIAl/view?usp=share_link

paulius2k, 2023-08-04:
> > Is there a compiled binary with #193 fix? I can't compile myself..
> 
> I uploaded it to my Drive: https://drive.google.com/file/d/1uxBnz5NtyNye1kTVE_dXRqnkUNpsOIAl/view?usp=share_link

@donivan2004 - just something to note: I only upgraded fw now, and my MQTT settings got messed up: Port, User, Pssw are wrong strings.
Device log says:
notice: Connection to MQTT server failed, rc=-2

Updating them manually resolved the issue.
I will monitor now to see if reconnect to router works.
Thanks.

closed 2023-11-11 by github-actions[bot]