fashberg/WThermostatBeca #129: Capturing

mojtabavi, 2021-09-27:
Hi, i'm noob to this device 
i have BAC-6000 
only things that i want is capturing the codes that device sends over the network and create a script that can control the device with computer or raspberrypi.
is that possible?

Master-Rudi, 2021-09-27:
MQTT is the keyword!Flash with Tasmota and use a MQTT Broker to send the states.Am 27.09.21 um 08:57 schrieb Seyed Ali Mojtabavi
                
            
            
                
                    Von: "Seyed Ali Mojtabavi" ***@***.***>Datum: 27. September 2021An: "fashberg/WThermostatBeca" ***@***.***>Cc: "Subscribed" ***@***.***>Betreff: [fashberg/WThermostatBeca] Capturing (#129)
                
                
Hi, i'm noob to this device
i have BAC-6000
only things that i want is capturing the codes that device sends over the network and create a script that can control the device with computer or raspberrypi.
is that possible?

mojtabavi, 2021-09-27:
is there anyway to find TCP Communication Protocol without flashing the device?

closed 2022-03-25 by mojtabavi