fashberg/WThermostatBeca #131: HA autodiscovery not working.

stav242, 2021-10-12:
i have a BHT-002GALW, configured with firmware WThermostat 1.19.beta1-fas (i have also tried the previous 1.18 version) . 

I can see mqtt messages, but the thermostat is not being auto-discovered in HA using mqtt.

do you have any suggested troubleshooting i can do ?

closed 2021-11-24 by stav242

jorch17b, 2021-11-28:
@stav242 have you solved the problem?

stav242, 2021-11-29:
Yes. Although can't pin down exactly which of the changes made it work.
Think it was my mqtt broker auto discovery setting that wasn't working. I
reinstalled the mqtt broker and reset all the config for it.

stav242, 2021-12-13:
On a side note....

I think I had a problem with the thermostat name having a # symbol in it.  I'll see if i can replicate it again.