fashberg/WThermostatBeca #135: cannot set target temperature Matt in iobroker

reutli, 2021-11-14:
I set up the mqtt-server in iobroker and I'm able to read alle the values & states.
Bat I'm not able zoom set a target temperature. If I do so, this has no effect against the device and will be overwritten after a few second.

So how should I'm able to set the desired temperature via iobroker?

ATBERKDA, 2021-11-14:
I‘ve the same issue but no solution until now. Tested with SonoffMQTT and MQTT-Server plug-in

langsoftware, 2022-09-13:
I have also the same issue, will this issue be fixed soon?

Bettman66, 2022-10-07:
This can be done without any problems in ioBroker, you only have to create an object yourself.
Example:
Wohnzimmer/stat/things/thermostat/properties/targetTemperature
Set:
Wohnzimmer.cmnd.things.thermostat.properties.targetTemperature

closed 2023-01-12 by github-actions[bot]