fashberg/WThermostatBeca #140: WiFi PSK limited / stripped to 32 chars?

SideshowBob79, 2021-12-26:
Hi Folke,

just tried your version of ThermostatBeca and it seems as if at least the web interface strips the WiFi PSK to 32 chars.
Since I use 63 I cannot use it right now and I couldn't find a workaround to inject the PSK in another way.
For sure this is a minor topic but for me it is blocking so I stick with Klaus' version now.
If 63 chars are supported, I will also test your firmware - I use ME81H thermostats.

best - and thank you for sharing your firmware with dummies like me that are only good for testing :o)

\Markus

Fran6u, 2021-12-27:
@SideshowBob79 Hi Markus, do you currently use the control in HA, if so, can you advise me on how you pulled the control into HA, and how I can do same..

SideshowBob79, 2021-12-28:
Hi @Fran6u ,
I do not use HA right now. I access the thermostat via MQTT / Node Red only.
\Markus

stav242, 2021-12-28:
Have you tried here?

https://github.com/fashberg/WThermostatBeca#integration-in-home-assistant

closed 2022-10-08 by github-actions[bot]