fashberg/WThermostatBeca #141: HA Autodiscovery not working

giovi321, 2021-12-29:
I'm not sure what I am doing wrong but HA auto discovery is not working at all. What are the steps to follow to enable it? Maybe I'm missing something.

paulius2k, 2022-01-02:
> I'm not sure what I am doing wrong but HA auto discovery is not working at all. What are the steps to follow to enable it? Maybe I'm missing something.

Have you followed _everything_ listed here?:
https://github.com/fashberg/WThermostatBeca#hass-autodiscovery

You have to see your new device in HA Configuration-->Integrations under the MQTT integration:
[image]

pfunkowski, 2022-01-08:
i have similar problem, finally problem is in mqtt password - firmware limit password length to 10 characters... standard mosquito password is too long ...
can You try
homeassistant > configuration >addin > mosquitto broker > configuration and add login like:

logins:
  - username: user
    password: passwd

restart mqtt broker and use Your new login in thermostat

closed 2022-10-08 by github-actions[bot]