fashberg/WThermostatBeca #144: Beca BHT-1000 device could not be reached / no ip adress / wifi

SirWouls, 2022-01-20:
So I flashed succesfully with firmware 1.18 and then tried to undo this going back to another firmware.
Now my devices ip address is somewhat lost (used to be able to connect via 192.x.x.x/config). 

Is there a way to get the thermostat back into wifi connection (by holding the down button)  to be able to redo the flashing?
Once I try to ./start_flash.sh again nothing can be reached so I guess my wifi adapter might be corrupted?

Any advice would be appreciated thanks.

Fran6u, 2022-01-20:
Your device probably got assigned a different ip. So, Search your WiFi manager for the new ip.

SirWouls, 2022-01-21:
> Your device probably got assigned a different ip. So, Search your WiFi manager for the new ip.

I checked that already and unfortunately that's not the case.

fashberg, 2022-10-02:
if flashing Over the Air went wrong you need to flash directly (see Flashing.md)

If it's just not connecting your device is either an Access Point or you can use the DOWN-button to go to recovery mode, but this requires a working firmware!

closed 2022-10-02 by fashberg