fashberg/WThermostatBeca #146: mqtt options deprecated in Home Assistant 2022.03

Flopescu, 2022-03-04:
Hi @fashberg 
First, a big thank you for your work, I have flashed my thermostat about 7-8 months ago and everything just works.

If you have some time, can you please have a look at the mqtt payload for home assistant?

In the version 2022.03 we have the following warnings about options that are deprecated:

```
Logger: homeassistant.components.mqtt.climate
Source: helpers/config_validation.py:817 
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), 
[issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22)) 
First occurred: 1:58:33 PM (5 occurrences) 
Last logged: 1:58:33 PM

The 'send_if_off' option is deprecated, please remove it from your configuration
The 'away_mode_command_topic' option is deprecated, please remove it from your configuration
The 'away_mode_state_template' option is deprecated, please remove it from your configuration
The 'away_mode_state_topic' option is deprecated, please remove it from your configuration
The 'hold_modes' option is deprecated, please remove it from your configuration
```

Appreciated!

titiviking, 2022-07-04:
@jressel01 has a working patch for this ( https://github.com/jressel01/WThermostatBeca/tree/patch-1 ).
I compiled his code and it works well on my BHT-6000.

Flopescu, 2022-07-05:
thanks! 
I've managed to compile it myself.

I had the same issue withe the HA Autodiscovery - managed to get it working after I've saved the "thermostat config" page...

clemente3905, 2022-09-09:
Hi guys, I'm a total noob. My thermostat are no more working with the 2022.9 update.
Can someone please point me out on a noob guide on how to get the @jressel01 working patch, compile and send to the thermostat?
I'm on a windows machine and do not have any idea on how to compile sources

labodj, 2022-09-09:
@clemente3905 if you want you can try my release https://github.com/labodj/WThermostatBeca/releases

fashberg, 2022-09-11:
Fixed in https://github.com/fashberg/WThermostatBeca/releases/tag/v1.20-fas 

Thanks you for feedback and contributions

closed 2022-09-11 by fashberg