fashberg/WThermostatBeca #150: Unable to compile code

titiviking, 2022-04-25:
Attempting to compile the code myself, the WAdapter renders errors:

```
Compiling .pioenvs\wthermostat\libfc6\ESPAsyncTCP-esphome\ESPAsyncTCPbuffer.cpp.o
Compiling .pioenvs\wthermostat\libfc6\ESPAsyncTCP-esphome\SyncClient.cpp.o
In file included from WThermostat\../lib/WAdapter/WAdapter/WNetwork.h:27,
                 from WThermostat\WThermostat.cpp:3:
WThermostat\../lib/WAdapter/WAdapter/WJsonParser.h: In member function 'WProperty* WJsonParser::endTrue()':
WThermostat\../lib/WAdapter/WAdapter/WJsonParser.h:500:23: error: cannot convert 'bool' to 'WProperty*' in initialization
  500 |   WProperty* result = false;
      |                       ^~~~~
WThermostat\../lib/WAdapter/WAdapter/WJsonParser.h: In member function 'WProperty* WJsonParser::endFalse()':
WThermostat\../lib/WAdapter/WAdapter/WJsonParser.h:514:23: error: cannot convert 'bool' to 'WProperty*' in initialization
  514 |   WProperty* result = false;
      |                       ^~~~~
Compiling .pioenvs\wthermostat\libfc6\ESPAsyncTCP-esphome\tcp_axtls.c.o
Compiling .pioenvs\wthermostat\lib83f\Hash\Hash.cpp.o
*** [.pioenvs\wthermostat\src\WThermostat.cpp.o] Error 1
=========================================================== [FAILED] Took 8.40 seconds ===========================================================
```

JuniorLogol, 2022-06-29:
Same problem.
Can anyone help?
Thanks

titiviking, 2022-07-02:
I have forked your repo do be able to perform some changes if necessary.
Code compiles, but the Auto-Discovery does not seem to work... the MQTT message does not seem to be send out as I still see an older version of that auto-discovery message in my MQTT...

I do see the updates from the thermostat in the correct topics, but sending commands still seems to fail.
Any idea why the auto-discovery is not being sent ?

jressel01, 2022-07-02:
Home assistant auto discovery works. I have 10 climate and all work fine
I mod all bht002 with hardware mod

paulius2k, 2022-07-19:
@jressel01 - is there a compiled fw file I could download and install?  thanks.

jressel01, 2022-07-21:
@paulius2k [My Firmware](https://github.com/jressel01/WThermostatBeca/releases/download/1.20.beta1-jrs/firmware.bin)

paulius2k, 2022-07-21:
@jressel01 thank you.