fashberg/WThermostatBeca #152: Hangs up when switching off from HA

FreakyPady, 2022-08-06:
Hi,

i'm a bloody beginner in using a thermostat with Home Assistant. But i bought a BHT-002 and flashed successful to firmware WThermostat 1.18-fas.

After that i set up an Mosquitto broker on my raspberry PI and integrated that with the MQTT integration into Home Assistant. Everything works... set temperature, go to manual or auto mode - fine. But if i switch the thermostate OFF it shows in HA a temperature of -3 degrees and then it hangs - no more control from HA possible.
The device display has lighten up but doesn't show anything, no action if i try to push one of the buttons.
In Browser i can access the ESP page and reinit the device - but doesn't help. Same if i access the network config page and press "save config and reboot".
If i view the info-screen, i can see the problem:  MCU-Initialized: No

Is just my new hardware hanging in this case?

Let me also tell, that i have bought a new BHT-002 with the tuya-controller WB3S. I replaced this tuya chip and changed to an ESP8266 (have seen that [here](https://github.com/klausahrenberg/WThermostatBeca/issues/241))

FreakyPady, 2022-09-02:
Annex: I ordered a second BHT-002 from Aliexpress. It came with the correct controller, so i just flashed it with the same firmware (see above). This device is working correctly and does not show the hanging-fault like described above! Very interesting for me...!

Any idea what could be the problem?

dynamik-one, 2022-09-22:
Hi,

I have the same experience as FreakyPady. Screen freezes/MCU seems to disconnect. Only way to get them running again is to reflash with 1.25beta.
BHT-002-GBLW with ESP-12F 1.20-fas en 1.21-fas. Luckily i could web-reflash it with the 1.25beta that it was forked from and after cycled the power the screen came to life again. 
Will do some additional testing tonight. Seems that i have a pcb with 002-cms-wifi-and-modbus-20210528-V1 written on it.

dynamik-one, 2022-10-03:
I did some thinkering. And it seems that i can control the thermostat physically and do anything a like with it without it "crashing" The funny thing is that i just upgraded to the 1.22-fas version. And where it usually keeps on -3 until i remove power from the thermostat by using the circuitbreaker it now displays the correct temp values in HA but the display is "off" but i can't switch it on. What would be the best order in supplying debugging information? tnx in advance.

dynamik-one, 2022-10-03:
Ok. Using operation auto or heat within HA doesn't seem hang up the thermostat (I can still manipulate the temp setting and i can feel the IR heating switch on). Choosing the off option(choices are auto,heat,off) within the operation setting instanly kills it. The power button within HA also kills it.

FreakyPady, 2022-10-04:
Maybe theres a difference in the device that comes with the "WB3S"? I had these problems just with the one i got delivered with the WB3S! Maybe they changed the MCU software also, not only the WiFi-Controller?

The last weeks, I ordered additional devices via AliExpress. They had all the "good" TYWE3S chip. I flashed all of them with v1.18 and they are all working without any problems! 

At this point: Thanks to fashberg for this great project and work! For me it's very usefull!

closed 2023-01-08 by github-actions[bot]