fashberg/WThermostatBeca #153: MQTT Hold mode removed in HA 2022.9

clemente3905, 2022-09-09:
Hi all, with the last update (2022.9) of HA core, the hold mode has been removed.
As a result the thermostat integration in HA is no longer working.

Is there any workaround other than avoid upgrading home assistant?

labodj, 2022-09-09:
@clemente3905 if you want you can try my release [HERE](https://github.com/labodj/WThermostatBeca/releases)
It is compatible with 2022.9
Just send to the thermostat as a normal update at http://THERMOSTAT_IP/firmware?

clemente3905, 2022-09-09:
@labodj First of all THANKS!
Then, this will be the very first time that I have to upgrade a Thermostat.
Can you please guide me on what steps do I have to perform to have them again online and working?
I imagine something like
1. Download and save the .bin file on a PC
2. open in a browser http://THERMOSTAT_IP/firmware?new_firmware.bin
3. Logging on the termostat (which wifi? which User? which password?)
4. Re-configure the termostat as it was before
5. How Do I change the config.yaml in home assistant to have the thermostat work again? Just remove the away-mode?
6. What abaout the hold mode in which I usually select manual/scheduler/eco? Is it still working in 2022.9?

Thanks again for your help and forgive if I ask lots of questions, but would really love to avoid damage...

labodj, 2022-09-09:
@clemente3905

I am supposing you are running fashberg firmware on your thermostat.

Just open a web browser
Go to http://192.168.21.29/firmware?

Select the file, click "update the firmware"

Wait for reboot

Done

Nothing else to do

clemente3905, 2022-09-09:
Going to test right now.
Thanks 
Come back in any case with results

clemente3905, 2022-09-09:
No luck...
I flashed all thermostat but after upgrade HA to 2022.9 again they all went KO.
I then modified the config.yaml to remove the "hold" reference, but this time devices became unresponsive, I am no more able to change their state.

For instance, this is the part of the config.yaml relative to the thermostat

`climate: 
    - name: Corridoio
      unique_id: 1gf43d
      availability_topic: "corridoio/tele/LWT"
      payload_available: "Online"
      payload_not_available: "Offline"
      action_topic: "corridoio/stat/things/thermostat/properties/action"
      action_template: "{{value}}"
      temperature_command_topic: "corridoio/cmnd/things/thermostat/properties/targetTemperature"
      temperature_state_topic: "corridoio/stat/things/thermostat/properties/targetTemperature"
      temperature_state_template: "{{value}}"
      current_temperature_topic: "corridoio/stat/things/thermostat/properties/temperature"
      current_temperature_template: "{{value}}"
      mode_command_topic: "corridoio/cmnd/things/thermostat/properties/mode"
      mode_state_topic: "corridoio/stat/things/thermostat/properties/systemMode"
      mode_state_template: "{{value}}"
      fan_mode_command_topic: "corridoio/cmnd/things/thermostat/properties/fanMode"
      fan_mode_state_topic: "corridoio/stat/things/thermostat/properties/fanMode"
      fan_mode_state_template: "{{value}}"
      hold_command_topic: "corridoio/cmnd/things/thermostat/properties/holdState"
      hold_state_topic: "corridoio/stat/things/thermostat/properties/holdState"
      hold_state_template: "{{value}}"
      hold_modes: [ "scheduler", "manual","eco" ]
      payload_on: "ON"
      payload_off: "OFF"
      modes: [ "heat", "cool", "fan_only", "off" ]
      min_temp: 10
      max_temp: 35
      temp_step: 1
      precision: 0.5`

labodj, 2022-09-09:
@clemente3905 you don't need that on home assistant, this firmware supports autodiscovery.
So
- delete that part from home assistant config
- restart your home assistant to reload the new config
- go to http://Thermostat_IP/
- click on "configure network"
- check MQTT config
- select "Support Autodiscovery for Home Assistant using MQTT"
- click "save config and reboot"
- Home assistant should discover the "new" device

I'm using this way, so I can confirm it works as expected.

clemente3905, 2022-09-09:
There is something that is not working...
I made as you said but I get a new entity without valid value

[image]

If I go inside the MQTT info I can see that everything seems OK, except the fact that in the value template it doesn't use '{{value}}' for the template.

I remember that I had an issue with "standard" fasberg and I had to separate JSON in single argument to have them work.
It should be this?

[image]
If you look at message 14 you will notice that there is a } missing and also a warning at the beginning...

labodj, 2022-09-09:
@clemente3905 
and this is mine
[image]

The waring on log is normal, just disable logging, it's useless for you.
your mcuid properties is bugged.

It's not the firmware, it's some config wrong on your side or some leftover.
Try to clean(delete) MQTT topics, remove devices and configs from home assistant.

Reboot everything e retry.

clemente3905, 2022-09-09:
Ok, thanks for your patience and the help jacopo. I still don't know  how but I will buy you a beer for sure!
It is clear to me how to remove devices (simply comment the configuration.yaml since they are all manually configured).
It is NOT clear to me how to  clean or delete MQTT topics from HA or clean leftovers.
For the other "autodiscovered" climate devices, I will simply go to Settings->Devices and delete them one by one.

Is it correct?

fashberg, 2022-09-11:
he @clemente3905 , check out 1.21.

Hold-State is also deprecated. I've implemented Away and Scheduler/Auto with "Presets", just use the autoconfiguration, it works fine on my test-device

clemente3905, 2022-09-14:
Hi @fashberg . 
I used 1.21 in one of my thermostat.
As with @labodj beta 1.19, using autodiscovery the thermostat is not working. on my side.
Again, same version but using a manual configuration is working.
If it can be interested, I always have an error on the json file, specifically a } is missing and this destroy the JSON file.
My workaround is to send all values singularly and change the template in the yaml configuration to {{value}}

Don't know if this can be the issue but this solved my problem.

If you need me to post some additional log/info please tell me.

fashberg, 2022-09-15:
Hey @clemente3905 ,

Which HA version do you have?
Is there anything in the HA logs?

Probably it's my change in 1.12 with adding profiles. 
You can safely downgrade to 1.20 and test, if auto discover is working there.

clemente3905, 2022-09-15:
@fashberg 
I'm using HA core 2022.9.1 on a virtual machine.
I ALWAYS used your firmware for my BAC but NEVER worked if I try autodiscovery OR if I do NOT select single value in the JSON.
When I installed it at the beginning (it was the 1.19 beta fas), I realized that the "full" json is not valid, due to the missing } I show you before.
Following images is taken directly from info of the thermostat:

[image]

The MCU-ID map is missing the closing } so when compiling a full json file it is not valid and it is not parsed correctly.

Please note that the same problem appears even if I use manual configuration WITHOUT the "single value" in the json file and without manually modifying the template into {{value}} in the configuration.yaml, so IMHO it is NOT a problem in the autodiscovery.
I do not know if the problem is due to my specific item (but I have 5 identical BAC-002 and ALL of them show the same behaviour), but I'm not able to have it work in autodiscovery, because I do not get any control other than the preset mode (so no temperature, no fan control, no mode).
Do you think I will have to try 1.20?
Thanks for help and hope I can be useful to fix this bug

labodj, 2022-09-15:
@clemente3905 @fashberg 
I can confirm that 1.21 is working flawlessly in HA 2022.9.4 with autodiscovery.

Just a note, I always used "Send all properties also in separate MQTT messages"

clemente3905, 2022-09-17:
@fashberg @labodj 
If it can be useful, please find hereafter the diagnostic of the MQTT integration where I have 5 "manually integrated" thermostats and 1 autodiscovered (not working) thermostat. This one is called Salotto.
Be careful, I also have a manual integrated Salotto, which is correctly working.
Please note that I check both the autodiscovery and the "Send all properties also in separate MQTT messages"

[mqtt-52931e3cbd611ec5dcd2eaf18dc50d47-Salotto-78df34da27a3aa8ca58905f9843aefee.json.txt](https://github.com/fashberg/WThermostatBeca/files/9591935/mqtt-52931e3cbd611ec5dcd2eaf18dc50d47-Salotto-78df34da27a3aa8ca58905f9843aefee.json.txt)

fashberg, 2022-10-02:
it's was not the non-closing ``{`` , it was the quote (``"``) in the MCU ID. The MCUID broke json response.
Please check 1.22

commit 18d441c2 (2022-10-02): v1.22:

* Fixes invalid JSON responses/escaping (fixes #153)
* Fixes MQTT-Reconnect

Code Changes in dependant WAdapter Library
closed 2022-10-02 by fashberg