fashberg/WThermostatBeca #154: Ability to set a fixed IP

Master-Rudi, 2022-09-10:
Would be awesome if it would be possible to set a fixed IP. Like I can do it with Tasmota devices

iMike78, 2022-09-11:
You can set up a fix IP in your router by the MAC of your thermostat, which is the better solution as if you set the IP in your thermostat, but the router gives the IP to an other device, the thermostat will never connect to the wifi.

Master-Rudi, 2022-09-12:
I don´t agree with that. Giving a fixed IP directly to the device is a much better solution. The reason is quiet easy:
I have about 80 IoT devices in my smart home. Most of them have a fixed IP within the range 100 to 199. My router gives to the rest (smartphones, notebooks, eg) a non fixed IP within the DHCP range 2 to 99.
After changing the router for example, i can restart my smart home within minutes and not hours by reseting all new IPs to my ioBroker and shortcuts, bookmarks and scripts...

So maybe you can set my concern as a feature request maybe? Maybe somebody could help.

Thanks for your great work!

FreakyPady, 2022-10-04:
I would also like to be able using a fixed IP. This option would be very nice.

What I strongly(!) miss is a user authentication: For example, my childs should not be able to reach the device and change schedules or something else! But if they know the IP-address, they can change anything... Theres no secure :-(

Master-Rudi, 2022-10-18:
@fashberg : Is there any possibility to add this feature like it is in Tasmota for example?
Thanks

Master-Rudi, 2023-01-17:
Still an open issue, any possibility to implement thas?

N-b-dy, 2023-02-22:
I also use this firmware and am also very interested in the function. Therefore I have made the necessary adjustments in the repo WAdapter included via submodule and created a [PR](https://github.com/fashberg/WAdapter/pull/2). 

Hopefully there is a chance for a review by @fashberg, so that it can be included here.

Master-Rudi, 2023-03-31:
@N-b-dy 👍 
Any chance to share your firmware? @fashberg seems to be not so active anymore...

Master-Rudi, 2023-07-01:
Still need this feature… any possibility to get it?

N-b-dy, 2023-07-17:
Sorry for my very late response, kind of pretty busy weeks...
As @fashberg does not seem to be interested in merging my PR in the WAdapter project, I'll fork this project as well and upload the binary there. Could take until mid or end of next week, I'll leave a note here once I'm done.

Master-Rudi, 2023-07-18:
Awesome!! Thanks a lot!

N-b-dy, 2023-08-01:
Sorry folks, it took longer than expected. The code changes I've made some months ago are still present and after some struggeling with the submodules I could recompile it, ready to push. Before that I wanted to test it again with one thermostat... And I've successfully bricked it. I need to reprogram it with the serial connection and investigate the issue with a D1 Mini or so. 

The only difference I've noticed so far is that when I tested with the static IP a few months ago, the thermostat was freshly programmed and not yet configured. The one I successfully bricked was already configured with MQTT, NTP, etc..

closed 2023-11-05 by github-actions[bot]