fashberg/WThermostatBeca #156: Mqtt port change

dmitijmaslak, 2022-09-12:
Hello, please remove restriction of mqtt port...i have port 44444....but your firmware only 4 symbols....
Please fix this

fashberg, 2022-10-02:
Hey,
the base library WAdapter is saving port as string into the small flash, with fixed len to 4 chars.
You are right, port range is up to 65565, but changing this means that the config layout has to be upgraded. Maybe sometimes.
For now not changeable, except you fork. Easier to change port to something <=9999

Folke

dmitijmaslak, 2022-10-02:
I have already removed this limitation and recompiled the firmware. thank you )

closed 2022-10-02 by dmitijmaslak

dieterlind, 2022-11-06:
@dmitijmaslak : Can you give me a hint where you changed what to make 5 digits possible! Thanks!

dmitijmaslak, 2022-11-07:
> @dmitijmaslak : Can you give me a hint where you changed what to make 5 digits possible! Thanks!

Edit: lib\WAdapter\WNetwork.h
Search “mqttport” and change: 4 to you need, and 1883 - default port use…