fashberg/WThermostatBeca #157: Floor temperature is not taking into account

svarogSLASHc, 2022-09-13:
Hi! I have (BHT-002-GxLW), in configuration enabled floor sensor. It works and the temperature is displayed in the mqtt. But as you can see from attached example it is still heating even if floor temp is 24.5 and target it 22, but air sensor temp is 21.5. Is it expected? How to configure to control heating relay by floor temp instead?
Thanks

`{
  "idx": "Sweet_home1",
  "ip": "192.168.2.210",
  "firmware": "1.19.beta1-fas",
  "switchBackToAuto": false,
  "temperature": 21.50,
  "targetTemperature": 22.00,
  "deviceOn": true,
  "schedulesMode": "off",
  "holdState": "manual",
  "ecoMode": false,
  "locked": true,
  "floorTemperature": 24.50,
  "mode": "heat",
  "action": "heating",
  "state": "heating",
  "mcuId": "IAYz2WK1th0cMLmL1.0.0"
}`

chipmoca, 2022-09-15:
Привет! Я думаю, надо по инструкции в самом BHT-002, выбрать по какому датчику (пола) отрабатывать.

fashberg, 2022-09-15:
Yes, controlling logic is done by MCU, see my README.md and thermostat docs (i have a copy in /docs)

closed 2022-09-15 by fashberg