fashberg/WThermostatBeca #159: Falsh BHT-6000 with CH340G

borkie, 2022-10-01:
Hi, has anyone been successful in flashing the thermostat BHT-600 (the round one) in question with CH340G adapter?
esptool always report this error:

`A fatal error occurred: Failed to connect to Espressif device: No serial data received`

I've tried a lot of time, with different baud, but without success.
I cabled the TYWE3S like this pic https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#22-connection-to-device-for-flashing

Has anyone had the same problem? how did you solve?

jj-uk, 2022-12-17:
Have you tried to flash the device with a different serial adaptor? Or are you able to supply the device with 3v3 externally (e.g. not from the serial adaptor) ?
Serial adaptors cannot supply much current and may not be able to power the circuity resulting in a failed flash.

catdogmaus, 2023-01-14:
I was successful only with unsoldering and soldering chip back after flashing. But I was in situation where the previous flash did not work and I needed to do it directly. If u still have OEM software on BHT-6000 u can use OTA with tuya convert. I have also done that successfully

closed 2023-04-21 by github-actions[bot]