fashberg/WThermostatBeca #162: schedulesMode changes from "auto" to "off" when a new target temperature is set

igtek, 2022-10-11:
Hello

I'm using WThermostat v1.22-fas on MOES BHT-002.
If a new target temperature is set by a mqtt command while the thermostat is on "auto" schedule mode, the property schedulesMode automatically changes to "off", but the thermostat schedule mode remains on "auto".

[image]

schedulesMode property is changed to "off" but the thermostat is always on "auto"

igtek, 2022-10-12:
I've checked the firmware version 1.20n and this issue does not occur: it runs correctly.

[image]

closed 2023-01-16 by github-actions[bot]

labodj, 2026-11-08:
@igtek I maintain a fork of this project and have released [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo). It fixes the missing manual-mode command when changing the target from Home Assistant/MQTT while scheduling is active. The requested temperature is saved first, then the firmware sends the manual-mode frame followed by the new setpoint frame. This keeps the ESP's reported mode and the command sent to the thermostat MCU consistent.