fashberg/WThermostatBeca #166: Accuracy of MCU built in controller on BHT002

Funkenschuhster, 2022-10-31:
The accuracy of the MCU seems to be a two-possition controller with a hysteresis of +/-0.5K. But for my case thats not a good controller setup. My floor heating takes 4-6h to heat up/cool down. So at setpoint of 22°C I cycle around 20.5 to 23°C.
A PI controller with PWM would be better choice, like tasmota thermostat. But I know thats in the hand of the MCU.
Is there any documentation about the MCU and may a way to flash it? Would be nice, if the control is done by ESP and MCU manages only relay and display.

roesl, 2022-12-12:
I have the same problem - with setpoint 20 the temp cycle around 18 to 22 - this ist unusable for floor heating.
Is it possible to get the temperature from an external sensor, because the internal sensor will be heated from the Thrermostat itself and measures the case temperature and not the room temperature

juhjar, 2023-01-15:
Yes, hysteresis and sensor offset is needed. 

In my case, there is 1,5deg difference between mounted or hanging on cable. Moving sensor outside of casing could help, but offset would be accurate enough.

[image]

Funkenschuhster, 2023-01-15:
You can Set an offset by pressing and hold the "Windows" key and clock for
5 seconds. Then change parameter 1.

Sent from my mobile device / Von meinem Mobilgerät gesendet

roesl, 2023-01-15:
Hello, How to change hysteresis? --Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.

kirscheGIT, 2023-01-15:
This should be parameter 2
[image]

roesl, 2023-01-15:
Thanks, but I would need a lower hysteresis than 1--Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.

kirscheGIT, 2023-01-15:
Better than 1? Looking at the behavior of my Beca, 1 means the hysteris span. 
So when you set the thermostat to 20 it will stop heating at 20.5 and start heating at 19.5. 
Better resolution can‘t be achieved as the measurement and set step size is only 0.5

juhjar, 2023-01-15:
Hmm, even I moved thermistor outside of casing... there is still 1,5deg difference on measurement, between mounted/hanging. Maybe power supply keeps casing too warm and room has to cool down a lot compared to setpoint, which makes it feel like hysteresis is too big.

On picture you can see two times when I let thermostat to hang on it's cable and mount it back. Could some one verify this too? Next I will drill some air vents on top of casing,

[image]

roesl, 2023-01-15:
I also moved the thermistor out of casing, and this works much better. The internal electronic sometimes heats the thermistor, so the difference between real room temperature and measured temperature is sometimes more than 3 when there is no ventilation when nobody is at home and then its cold when coming home 😭--Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.

kirscheGIT, 2023-01-15:
On my Beca it sits here and it works well enough for me.

[image]

What you could do, is to have a separate thermometer in the room and use this value to manipulate the set temperature and turn the thermostat on and off. This can be done with openHAB at example.

jwillmer, 2023-01-16:
> What you could do, is to have a separate thermometer in the room and use this value to manipulate the set temperature and turn the thermostat on and off. This can be done with openHAB at example.

I already asked for something like this: https://github.com/fashberg/WThermostatBeca/issues/173

closed 2023-04-23 by github-actions[bot]