fashberg/WThermostatBeca #171: MOES BHT-002GBLW now with CB3S

mini-mi, 2022-11-27:
Hi,

i bought 4 BHT-002GBLW from MOES and have the problem that they are now using the new CB3S chip.
So it is not possible to use this anymore or am i wrong?

I try to buy other and hope to get some with the old chip or try to change the chip.

Regards

Max

rande83, 2022-12-02:
Hi.

It is still possible with soldering skills....

I replaced the CB3S with an ESP-12S.

btk999, 2023-01-02:
> Hi.
> 
> It is still possible with soldering skills....
> 
> I replaced the CB3S with an ESP-12S.

is this a 1:1 replacement of the chip or are there any other things necessary?

rande83, 2023-01-02:
> > Hi.
> > 
> > It is still possible with soldering skills....
> > 
> > I replaced the CB3S with an ESP-12S.
> 
> is this a 1:1 replacement of the chip or are there any other things necessary?

Hi.

Yes the ESP-12S is 1:1 compatible.

With other ESP boards you may have problems booting, even if the pinout is identical. This is because certain pins still need a pull up and pull down resistor.

So better take the ESP-12S directly, flash it before soldering and then replace the CB3S with it.

Small hint: When soldering better remove the PCB from the display, otherwise you will get permanent black spots on the display due to the heat.

btk999, 2023-01-02:
> Hi.
>
> Yes the ESP-12S is 1:1 compatible.
>
> With other ESP boards you may have problems booting, even if the pinout is identical. This is because certain pins still need a pull up and pull down [...]

Thanks! So no additional shortage Connections necessary. Im gonna try it :)

rande83, 2023-01-02:
> 
> Thanks! So no additional shortage Connections necessary. Im gonna try it :)

No.
Only if you want to do the hardware hack to connect the heating relay to gpio5.

btk999, 2023-01-02:
What is the advantage oder that?

rande83, 2023-01-02:
> What is the advantage oder that?

The advantage is that u can see the real status of the heating relay in the web interface and via MQTT, I guess. 

[image]

btk999, 2023-01-02:
Thanks! I searched for it but i could Not find further information, is Thais documentated somewhere?

rande83, 2023-01-02:
> Thanks! I searched for it but i could Not find further information, is Thais documentated somewhere?

[image]

It is mentioned in the configuration.md

And further in this link: https://github.com/klausahrenberg/WThermostatBeca/issues/17

mini-mi, 2023-01-02:
I can confirm it works with this ESP-12F

https://www.amazon.de/gp/product/B0B143QLPW/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1

And I also recomment to work careful and slowly otherwise you can damage the display, but 3/4 worked well.

timeo-78, 2023-03-17:
Did anyone check if the BHT-002 GB B (without wifi) has the same PCB without the ESP module? In this case it would be easy to add ESP-12S or ESP-12F without unsoldering the CB3S chip. Just a thought I have, wondering if someone already checked it.

closed 2023-06-22 by github-actions[bot]