fashberg/WThermostatBeca #181: Cat access the gui after first setup.

sasaderer, 2023-01-04:
Hi!

After flashing the ME81H device I was able to access the gui - I entered my Wifi network and MQTT broker info.
It connected to wifi and is sending some data to mqtt broker, but when I access the device with web browser, the "root page" does not show the GUI anymore but instead this: 

[image]
Is it possible to somehow to access the GUI again or reset the device via MQTT commands?

Br

kirscheGIT, 2023-01-04:
Append /config? to the ip in the browser.

sasaderer, 2023-01-04:
Wow - so it was that simple - I do see the GUI there:)
Thank you!

closed 2023-04-10 by github-actions[bot]