fashberg/WThermostatBeca #182: Question: is there a way to flash WB3S chipset?

shakin89, 2023-01-10:
Hi @fashberg,
i would like to use your firmware on my thermostats.
At the moment i own 3 Beca Smart BAC-002-ELW thermostats and need to buy other 6.
I saw a video from you, where you flashed with tuya convert a beca smart thermostat.
After opening mine thermostats, i've discovered they are equipped with a KWH-WB3S chipset.
After a little research i've discovered that wb3s chipset are not directly supported by tuya convert, but there is a port from libretuya that supports this chipset (as i understand https://docs.libretuya.ml/boards/wb3s/).
Do you think there will be a way to flash libretuya and then this firmware on wb3s chipst o pheraps esphome so they can integrate with home assistant?
Thanks in advance.

p.s. at the moment i have the thermostats functioning with local-tuya, but they are still connected to the cloud.

kuba2k2, 2023-03-18:
For disconnecting WB3S modules from Tuya, you can use tuya-cloudcutter, which is an equivalent of tuya-convert.
The LibreTuya firmware is partially compatible with ESP32/ESP8266 libraries, so with some adaptations it should be possible to compile this repository on WB3S. (Disclaimer: I'm the author of LibreTuya)

shakin89, 2023-04-09:
Wow, first of all thanks for your great work and for your suggestion. 
I will try to use cloudcutter or to flash libretuya when i will stop using my thermostats... :) 
In the meanwhile i will study the libretuya functionality.

closed 2023-04-09 by shakin89