fashberg/WThermostatBeca #186: WIFI access dead

jwillmer, 2023-01-21:
My AP was down over the night and since then one of my BHT-002 does not reconnect anymore. All others are fine. I switched out the panel with another and it connects immediately. No idea what could cause this issue. I have the WIFI fallback activated in the web portal but the device wont spin up an access point - my others do. No idea what happened or how I can fix it. Do I have to re-flash (unsolder) it?

Deepmaster10, 2023-01-22:
No, you need to turn off and then turn on the power to the thermostat

jwillmer, 2023-01-22:
It does not help. As I said, when connecting a different display it works out of the box. So it has to do with the unit in the display and not the high voltage part.

Deepmaster10, 2023-01-22:
View receives the IP address of the thermostat in the web router? The
panel cannot influence, it works with the CPU. The esp8266 module is
responsible for wifi. Panel problem! Without a panel, WiFI should
connect. I correctly understood, the touch panel?

jwillmer, 2023-01-22:
I check my router DHCP table as well as the connected devices of my WIFI AP and the thermostat does not show up - I know the MAC of it so I am sure it does not show up. It also does not create any new network I could connect to (the setting is enabled in case it can't reach the WIFI). When I replace the detachable part of the thermostat that houses the MCU and the ESP8266 with another unit it connects immediately to my WIFI. So the bottom part of the thermostat seams not to hav any problem. The defect thermostat also works without issues it just seams to not connect to any WIFI.

Deepmaster10, 2023-01-22:
Network Recovery

If you cannot access your device, try to switch to Access-Point mode:

   - Power off the device by Pressing the button in the middle.
   - Then press the "down" button (most right) for about 8 seconds.
   - The Screen starts to blink and shows a WiFi-Icon.
   - The thermostat is now an Access Point named Thermostat-xxxxxx. Default
   password is 12345678
   - Now you can fix network settings.
   - Saving settings or pressing power button switches back to Station mode

If that doesn't help, you'll have to try flashing... or changing the
ESP8266 module.

jwillmer, 2023-01-22:
Yes, I'm not able to get any reaction when pressing the button for 8+ sec. Have to reflash -.-

closed 2023-01-22 by jwillmer

PaoloTK, 2023-04-05:
Same problem here. Any idea what caused the issue? Don't fancy reflashing every few months.