fashberg/WThermostatBeca #187: Tuya/Beca BHT-1000 Thermostat works perfectly, even the Hardware hack

hakspiel, 2023-01-26:
I tried my luck with a newly purchased Beca BHT-1000 (Production 07/2022).

I replaced the original CB3S WiFi Modul with an ESP8266-12F .
That is a 1:1 replacement, Pin Compatible.

What shall i say, everything works on the first try.

Hardware hack also works.

I am happy, thank you for your Project!

Original BHT-1000:
[image]

With replaced ESP-12F Module:
[image]

With Hardware-Hack (next one will look better, but works):
[image]

malgates, 2023-01-31:
Nice one! What does the Hardwarehack do?

hakspiel, 2023-02-02:
The problem without Hardware hack is that you cannot detect the relais position. There is only an assumption available. 

The hardware hack connects the 5v output (that drives the relais) to the input of the ESP (you need the voltage devider otherwise ESP might be grilled with 5V). 
Now the ESP can directly detect the position of the relais. 

This is implemented in the Software, selectable via dropdown Menu in the configuration.

closed 2023-05-08 by github-actions[bot]