fashberg/WThermostatBeca #196: Basic Auth for Webinterface

msebald, 2023-03-26:
I want to come back to #122 as I have the same request. Before people (e.g. my kids ;-) ) on the home network start to scan and modify stuff, I want to make sure that everything has some basic protection. Like http auth.

From Tasmota I know this feature. So it should be here, too. Is there any way to get onto the console? If yes, I can enable it myself, along with other Tasmota features.

msebald, 2023-06-25:
This feature would still be great and is a huge improvement.

closed 2023-12-31 by github-actions[bot]

labodj, 2026-10-27:
@msebald I’ve added optional HTTP Digest authentication in my [maintenance fork](https://github.com/labodj/WThermostatBeca-labo).

Administrator login is enabled by default and uses admin with the saved Wi-Fi password. It protects web administration, firmware uploads, WebThings writes and WebSocket logs. The requirement can be disabled explicitly in Network Configuration; MQTT keeps its separate broker credentials.

No configuration reset is required for supported legacy settings. HTTP traffic is still unencrypted, so administration should remain on a trusted network. The README explains the access and recovery behavior.