fashberg/WThermostatBeca #205: Least Home assistant doesn't like MQTT format

ardenking, 2023-08-03:
After updating to latest HA get this following warning about the MQTT formatting

any advice on resolving 

[image]

ardenking, 2023-08-03:
esphome:
  name: in-therm-pwt-002-1
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: 600
    then:
      - switch.turn_off: power_switch

# Make sure logging is not using the serial port
logger:
  baud_rate: 0
  
# Enable Home Assistant API
api:

ota:
  password: "YOUR_OTA_PASSWORD"

wifi:
  ssid: "YOUR_WIFI_SSID"
  password: "YOUR_WIFI_PASSWORD"
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "In-Therm-Pwt-002-1"
    password: "YOUR_GENERATED_PASSWORD"

captive_portal:

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Getting time from HA
time:
  - platform: homeassistant
    id: ha_time

# Register the Tuya MCU connection
tuya:
  time_id: ha_time

climate:
  - platform: tuya
    name: "Thermostat"
    switch_datapoint: 1
    target_temperature_datapoint: 2
    current_temperature_datapoint: 3
    eco_datapoint: 5
    eco_temperature: 20 °C
    temperature_multiplier: 0.5
    visual:
      min_temperature: 20 °C
      max_temperature: 35 °C
      temperature_step: 0.5 °C

# Create a sensor
sensor:
  - platform: "tuya"
    name: "Temperature"
    sensor_datapoint: 3
    unit_of_measurement: "°C"
    device_class: "temperature"
    accuracy_decimals: 1
    filters:
      - multiply: 0.5

switch:
  - platform: "tuya"
    name: "Power"
    id: power_switch
    switch_datapoint: 1
    internal: true
  - platform: "tuya"
    name: "Eco Mode"
    icon: "mdi:earth"
    switch_datapoint: 5
    internal: true
  - platform: "tuya"
    name: "Lock"
    icon: "mdi:lock"
    switch_datapoint: 6
    internal: true

ardenking, 2023-08-03:
this should be enough to get you started

FreakyPady, 2023-08-09:
Wow, interesting... i found this topic because i got the same message in Home Assistant now after updating to 2023.8.

I'm not pretty sure if i understand right what you described here, ardenking. I'm also using ESPhome addon in Home Assistant, but with other devices not with my thermostat (BHT-002 with WThermostat 1.18-fas at the moment). 
Do I understand right, that it's possible to flash the ESP's inside just with ESPhome and adding your code to control them simply with ESPhome via Home Assistant? 

In the past i didn't ever thought about the possibility with ESPhome - i just flashed the described file and configured the thermostat via webinterface... But this could be a possible solution, if 2024.2 will come bevor an fix of the described error is available.

ardenking, 2023-08-09:
Yep I put the yaml in esphome addon downloaded the bin file and updated via the web interface on this firmware

FreakyPady, 2023-08-10:
Thanks, that's really good to know. After that, the device should be able to get updates via ESPhome OTA?

Maybe I will test this, just for fun. In comparison to the actual used software i just would miss the possibility of the time schedule: Changing the integrated schedule via webinterface is very comfortable. If using your yaml via ESPhome i need to change the time schedule directly at every device or let HA doing all the changes, right?

ardenking, 2023-08-10:
Im using the scheduler from HACs to set timers directly in Lovelace Its made things better because its all in one place

MJP-76, 2023-08-10:
Hope I am not hijacking the thread here but thought I would pop in my 2cents worth

Been using ESPHome on mine from day 1 and use a scheduler to run them. I have 3 for separate underfloor heating and since its ESPHome it uses the esphome integration and not mqtt

As I get the floor temp into home assistant it manages them based on time of day, year, who's home etc etc along with all my other heating as I use schedy via Appdaemon to control all my heating

All you should need to change in my code for these devices to appear in home assistant is install the ESPHome addon and then change the "substitutions" with your device name, set up the sections in your esphome secrets file for wifi, ha API key etc, and generate the .bin and upload to your thermostat

There are many other little tweaks you can add like turning off the relay/switch when power is restored etc comes on etc

One thing I like is the restart if API connection is lost which if your HA goes down, it will reboot and automatically "power off" so no need to worry if 

Here is my ESPHome code:

`substitutions:
  devicename: Entrance UFH
  deviceid: entranceufh

esphome:
  name: ${deviceid}
  platform: ESP8266
  board: esp01_1m

logger:
  baud_rate: 0

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${devicename}_AP
    password: !secret ap_password
  domain: !secret domain
  fast_connect: true

captive_portal:

api:
  reboot_timeout: 15min
  encryption:
    key: !secret api_key

ota:
  password: !secret ota_password

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

web_server:
  auth:
    username: admin
    password: !secret web_server_password

time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ${devicename} IP Address

binary_sensor:
  - platform: status
    name: ${devicename} Status

tuya:
  time_id: homeassistant_time

climate:
  - platform: tuya
    name: "${devicename}"
    switch_datapoint: 1
    target_temperature_datapoint: 2
    current_temperature_datapoint: 102
    eco_datapoint: 5
    eco_temperature: 16 °C
    temperature_multiplier: 0.5
    visual:
      min_temperature: 15 °C
      max_temperature: 23 °C
      temperature_step: 0.5 °C

sensor:
  - platform: wifi_signal
    name: "${devicename} WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "${devicename} Uptime"
  - platform: "tuya"
    name: "${devicename} Target Temp"
    sensor_datapoint: 2
    unit_of_measurement: "°C"
    device_class: "temperature"
    accuracy_decimals: 1
    filters:
      - multiply: 0.5
  - platform: "tuya"
    name: "${devicename} Room Temp"
    sensor_datapoint: 3
    unit_of_measurement: "°C"
    device_class: "temperature"
    accuracy_decimals: 1
    filters:
      - multiply: 0.5
  - platform: "tuya"
    name: "${devicename} Floor Temp"
    sensor_datapoint: 102
    unit_of_measurement: "°C"
    device_class: "temperature"
    accuracy_decimals: 1
    filters:
      - multiply: 0.5
  - platform: "tuya"
    name: "${devicename} Mode"
    sensor_datapoint: 4
    internal: true

switch:
  - platform: restart
    name: "${devicename} Restart"
  - platform: "tuya"
    name: "${devicename} Power"
    id: power_switch
    switch_datapoint: 1
  - platform: "tuya"
    name: "${devicename} ECO Mode"
    icon: "mdi:earth"
    switch_datapoint: 5
  - platform: "tuya"
    name: "${devicename} Child Lock"
    icon: "mdi:lock"
    switch_datapoint: 6
  - platform: "tuya"
    name: "${devicename} DP 104"
    switch_datapoint: 104
    internal: true
`

ardenking, 2023-08-10:
Thanks for the input hopefully between my yaml and yours should be enough for others to roll their own for their environment

MJP-76, 2023-08-10:
Used the code button plenty of times, not sure why it just keeps formatting as it does

Esphome has a nice add-on that creates a dashboard where you can edit and update you config / firmware easily without leaving home assistant

[image]

Red devices are offline, grey are up to date and then a blue update if there is a new firmware ready to go and a nice button that says "Update All" which i nice when you have lots of esphome devices as it will build and update for you one at a time

I think esphome you can set home assistant to do a ping of the devices but cant remember, this will shows the online/offline state

For each device, you can manually update, check logs etc or click the visit button which takes you to the web interface of the device where you can control it, see logs and run updates etc as below

[image]

I found that if you have an "internal" domain name on your router setup then its easier to get to the devices if HA is down by just typing the device name.domain.name etc but if not, just the device name should service on a normal ISP router

I would highly recommend the settings I have for wifi, api, captive_portal  and web_server as a bare minimum

That way if you ever lose connectivity to HA or even the wifi SSID they are configured for, they restart with their own AP

Hope that answers your questions

And just a bit on my heating:

My heating is 373 lines of code, it controls 13 radiators, 5 ceiling fans, 2 air cons and monitors about 14 room temperatures and 1 external temperature. It switches to different schedules based on the external temp i.e Summer/winter 

If I am away the wife generally migrates to the bedroom when I am, so why heat the house or if we are both away it effectively shuts it down so it only protects against condensation etc (In the UK)

My son works shifts so why heat his room when not there

All done with Home Assistant, a few sonoff TH16 on my heaters, and some aqua sensors :)

MJP-76, 2023-08-10:
Oh, One thing I forget, Sometimes when you have "other" firmware like tasmota on the devices, there is not enough memory to load the esphome

I have even had it with an old device with a much older esphome on it that you need to load the sonoff_mini firmware I think it is or as you can see, i have a very, very basic esphome config that fits and once its running you can upload the new config

So, best go with a minimal firmware, the put your full one on it if you are changing over

FreakyPady, 2023-08-10:
Thanks for your very detailed report. I am also using the ESPhome addon in HA, but just played around a bit... I build an own counter on my gas meter with this, very simple! But didn't ever thought about the possibility of changing my thermostat to ESPhome platform.

In my home, i'm having 8 thermostate running underfloor heatings (water, not electric). Because of that, heating is very slow... Heating must start for example 2 hours before somebody comes home and can stop 2h before the last one leaves. Because of that, i'm having an minimum time schedule directly in every thermostat. In additional to that, HA controls all the exceptions: School holidays, holidays of me or my wife, ... Presence would be detected also, but not to start the heating. It's just for the decision to stop heating or not! 

Because of that, for me it's usefull to use an time schedule instead of a presence detection. And if my raspberry (running HA) will stop working (and i'm maybe not at home and my wife don't like to freeze) it's better to put this "minimal time schedule" directly in the thermostat. So, in the worst case, heating is independent from HA.

I saw that with the firmware from klaus ahrenberg it should be possible to change the integrated schedule via MQTT-command ( https://github.com/klausahrenberg/WThermostatBeca ). Do you know about the possibility? That should be also possible to integrate in ESPhome, if you have the knowledge...

hakspiel, 2023-08-25:
i made some changes to naming of the Climate Entity in the code. HA does not show this error message anymore.

pczolee, 2023-08-26:
> i made some changes to naming of the Climate Entity in the code. HA does not show this error message anymore.

Thank you! The binary in you repo is the final version? Can I upload and update from the web in 1.22?

hakspiel, 2023-08-26:
it is based on latest FAS version.

pczolee, 2023-08-26:
> i made some changes to naming of the Climate Entity in the code. HA does not show this error message anymore.

Thank you! The binary in you repo is the final version? Can I upload and update from the web in 1.22?

> it is based on latest FAS version.

Yes it it clear. But I tried to update from the web, and the resoult is this:
[image]

So, something is wrong...

hakspiel, 2023-08-26:
for my thermostat it worked, when i was flashing over FAS 1.22. Can you descride your configuration?

pczolee, 2023-08-26:
> i made some changes to naming of the Climate Entity in the code. HA does not show this error message anymore.

Thank you! The binary in you repo is the final version? Can I upload and update from the web in 1.22?

> it is based on latest FAS version.

Yes it it clear. But I tried to update from the web, and the resoult is this:
[image]

So, something is wrong...

> for my thermostat it worked, when i was flashing over FAS 1.22. Can you descride your configuration?

Nothing special in the config. I have more but everyone have the same problem. Are there any way to export the config, or debug what happening?

andreas-bulling, 2023-10-22:
@MJP-76 Thanks a lot for your ESPHome example - I am trying to use it with a "Moes WHT-NV-GB-WH-M" thermostat and at first I thought it is all working fine. What's strange, however, is that activating the heading mode does not work entirely reliable. Is it possible that the thermostat only heats the floor if the target temperature is above the room temperature?

andreas-bulling, 2023-10-26:
@ardenking I have also tried your code (which is similar to the one from @MJP-76 but I thought I'd give it a try anyway.

The thermostat is working in principle. What's strange though is that when the target temperature is 20 degrees (the eco temp in my case) and the floor has  a temperature of, say, 16 degrees, then the thermostat will only switch the relay once I set the target to 24 degrees - below it'd wouldn't start heating.

Is that also true for you/do you observe the same behaviour? Or do you have any idea why that is?

andreas-bulling, 2023-10-26:
I think I found a solution. It seems the thermostat heats with respect to the room temperature (why? no idea - doesn't make sense in my opinion). But the thermostat luckily offers a "system function" menu through which internal settings can be accessed. One setting is about the internal and external temperature sensor. I've set that to "external only" and this seems to do the trick.

Bl00d-B0b, 2024-01-27:
Hello,

maybe you have answer for question above?
The "**Datapoint 104: switch**": Do you know about the function?

Device BH-002, default value is ON after reboot.

Ed

closed 2024-05-02 by github-actions[bot]