fashberg/WThermostatBeca #208: CB3S BK7231N

tomik671108, 2023-09-21:
Hi.
I purchased BHT-002-GCLW and I have a problem communicating with it.
Inside there is a different CB3S module than yours with the BK7231N chip.

[https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076](url)

Does anyone have experience with such a case?

openshwprojects, 2023-09-24:
[OpenBeken](https://github.com/openshwprojects/OpenBK7231T_App) supports TuyaMCU well in a scriptable and customizable manner, we can also have custom pages [hosted on OBK LittleFS](https://www.elektroda.com/rtvforum/topic3971355.html), so it should be possible to support it easily.

Kihltech, 2023-12-23:
I have an BHT-002GALW-W / BHT-002-GALW with this CB3S chip, which I'm interested in flashing to this firmware. 
From what I understand I'm not able to directly use the current firmware. 
I have a suitable ESP programmer and I can attempt to extract the firmware if that can be usable. I'm also willing to donate one unit to help with the reverse engineering. 

Here's a picture of the unit:
[image]

For future reference, I believe two resistors need to be removed to allow flashing:
https://www.elektroda.com/rtvforum/topic4003974.html

tomik671108, 2023-12-23:
Hi.
The latest TUYA devices are equipped with Beken chips, which cannot be loaded with Tasmota, only OpenBK.
Read this entire thread (and others about OpenBK):
[https://www.elektroda.com/rtvforum/topic4003974.html](url)
, see if you are able to backup the original firmware first and then flash OpenBK.
Instructions are in the thread.
Indeed, resistors Q1 and Q2 need to be desoldered during the first upload of OpenBK, later updates are OnAir.
At the same time, you can add an additional connection that will allow you to obtain information about the actual status of the relay.
When you are ready, I will show you my thermostat configuration in Home Assistant, and help you with the first configuration of OpenBK.

paravoid, 2024-01-09:
For what it's worth, besides OpenBK, you can nowadays flash the Beken chips with ESPHome.

I have a working config here:
https://devices.esphome.io/devices/Beca-Thermostat

Note that I did not have to desolder anything to flash, everything just worked.

closed 2024-04-14 by github-actions[bot]