fashberg/WThermostatBeca #226: Feature request: automatically turn on the thermostat when connected to power

giovi321, 2024-04-01:
Hi,
the BHT-6000 does not automatically turn on when power is given to it. It would be nice to have the option to turn on automatically the device when power is given to it, the equivalent of sending an mqtt message like this:
```
topic: thermostat1/cmnd/things/thermostat/properties/deviceOn
message: true
```
Thanks!

Stoatwblr, 2024-06-04:
Are you sure about that? At least some of these devices have this option in the presets (resume, off, on)

giovi321, 2024-06-04:
I'm quite sure... where should I see this option? Sending a MQTT message?

tomik671108, 2024-06-07:
I would also be interested in such a function, remembering the state before the power failure would be very helpful.

giovi321, 2024-06-27:
I checked, there's no preset {resume, off, on}.

`{"idx":"therm_1piano","ip":"192.168.165.156","firmware":"v1.22-fas","switchBackToAuto":false,"temperature":17.50,"targetTemperature":5.00,"deviceOn":true,"schedulesMode":"off","holdState":"manual","ecoMode":false,"locked":false,"mode":"heat","preset":"none","action":"idle","state":"off","mcuId":"IAYz2WK1th0cMLmL1.0.0"}`

closed 2024-10-01 by github-actions[bot]