fashberg/WThermostatBeca #236: Bug fixes and improvement recommendations

Gfermoto, 2025-05-21:
# Bug Fixes and Improvement Recommendations for WThermostatBeca

After analyzing the codebase, I've identified several issues and areas for improvement:

## 1. Unhandled Case in updateModeAndAction()

**Issue:**  
In `WBecaDevice.h` (~line 2160), there's an unhandled case for the `MODEL_BAC_002_ALW` when `systemMode` doesn't match any of the three listed modes (`SYSTEM_MODE_HEAT`, `SYSTEM_MODE_COOL`, `SYSTEM_MODE_FAN`).

**Fix Recommendation:**  
```cpp
} else if (getThermostatModel() == MODEL_BAC_002_ALW){
    if (this->systemMode->equalsString(SYSTEM_MODE_HEAT)){
        this->mode->setString(MODE_HEAT);
    } else if (this->systemMode->equalsString(SYSTEM_MODE_COOL)){
        this->mode->setString(MODE_COOL);
    } else if (this->systemMode->equalsString(SYSTEM_MODE_FAN)){
        this->mode->setString(MODE_FAN);
    } else {
        // Handle unknown mode: set COOL as default
        network->log()->warning(F("updateModeAndAction: Unknown system mode, defaulting to COOL"));
        this->systemMode->setString(SYSTEM_MODE_COOL);
        this->mode->setString(MODE_COOL);
    }
}
```

## 2. MCU Initialization Issue

**Issue:**  
In `WBecaDevice.h` (~line 1272), there's a workaround with the comment `FIXME HY08WE: no answer from MCU on Working Mode Query`.

**Fix Recommendations:**  

Option 1 - Add timeout:
```cpp
case 3:
    network->log()->notice(F("Query the MCU working mode."));
    {
        unsigned char mcuCommand[] = { 0x55, 0xAA, 0x00, 0x02, 0x00, 0x00 };
        commandCharsToSerial(6, mcuCommand);
    }
    mcuInitializeState++;
    mcuQueryTimeout = millis() + 3000; // 3 seconds timeout
    break;
case 4:
    /* wait for answer from step 3 */
    if (millis() > mcuQueryTimeout) {
        network->log()->warning(F("No answer from MCU on Working Mode Query, continuing anyway"));
        mcuInitializeState++; // continue even without response
    }
    break;
```

Option 2 - Model-specific handling:
```cpp
case 3:
    if (getThermostatModel() == MODEL_HY08WE) {
        // This model doesn't respond to working mode query
        network->log()->notice(F("Skipping working mode query for HY08WE"));
        mcuInitializeState = 5; // Skip steps 3 and 4
    } else {
        network->log()->notice(F("Query the MCU working mode."));
        {
            unsigned char mcuCommand[] = { 0x55, 0xAA, 0x00, 0x02, 0x00, 0x00 };
            commandCharsToSerial(6, mcuCommand);
        }
        mcuInitializeState++;
    }
    break;
```

## 3. NTP Synchronization Issues

**Issue:**  
In `WClock.h`, there are issues with NTP server not responding or time being considered invalid.

**Fix Recommendation:**  
Improve NTP error handling with retry mechanisms and alternative servers:

```cpp
// In WClock declare an array of NTP servers
const char* ntpServerFallbacks[] = {"pool.ntp.org", "time.google.com", "time.windows.com", "time.apple.com"};
uint8_t currentNtpServerIndex = 0;

// In the failed sync handling method
if (!ntpClient.update()) {
    network->log()->warning(F("NTP sync failed with %s. Trying alternative server."), ntpServer->c_str());
    
    // Switch to next server in the list
    currentNtpServerIndex = (currentNtpServerIndex + 1) % (sizeof(ntpServerFallbacks) / sizeof(ntpServerFallbacks[0]));
    String newServer = ntpServerFallbacks[currentNtpServerIndex];
    ntpClient.setPoolServerName(newServer.c_str());
    
    // Set shorter interval for next attempt
    lastTry = now - (ntpRetryMinutes * 60 * 1000) + 30000; // Try again in 30 seconds
}
```

## 4. Incomplete JS Template

**Issue:**  
In `WBecaDevice.h` (~line 144), there's a constant string with a TODO placeholder.

**Fix Recommendation:**  
Implement the required JavaScript code or remove the template if not used:

```cpp
const static char PAGE_BECA_JS[] PROGMEM = R"=====(
// Implementations of client-side scripts for device controls
function updateThermostatUI() {
    // Code to update UI elements based on thermostat state
}

function sendThermostatCommand(cmd, value) {
    // Code to send commands to thermostat
    fetch('/api/thermostat/' + cmd, {
        method: 'POST',
        body: JSON.stringify({value: value}),
        headers: {'Content-Type': 'application/json'}
    })
    .then(response => response.json())
    .then(data => updateThermostatUI(data));
}
)=====";
```

## 5. Serial Port Error Handling Improvement

**Recommendation:**  
Add error handling and timeouts for serial communication:

```cpp
void commandCharsToSerial(unsigned int length, unsigned char* command) {
    int chkSum = 0;
    bool serialSuccess = true;
    
    if (length > 2) {
        for (unsigned int i = 0; i < length; i++) {
            unsigned char chValue = command[i];
            chkSum += chValue;
            
            // Add write success check
            size_t bytesWritten = Serial.write((char) chValue);
            if (bytesWritten != 1) {
                serialSuccess = false;
                network->log()->error(F("Failed to write byte to serial"));
            }
        }
        
        unsigned char chValue = chkSum % 0x100;
        size_t bytesWritten = Serial.write((char) chValue);
        if (bytesWritten != 1) {
            serialSuccess = false;
            network->log()->error(F("Failed to write checksum to serial"));
        }
        
        if (serialSuccess) {
            network->log()->trace(F("commandCharsToSerial: %s, ChckSum 0x%02hhx"),
                getCommandAsString(length, command).c_str(), chValue);
        } else {
            network->log()->error(F("Serial communication error"));
        }
    }
}
```

## 6. General Recommendations for Code Improvement

1. **Add more detailed error logging**:
   - Log all unusual states and transitions
   - Log all MCU communications during debugging 

2. **Improve error handling and recovery**:
   - Add recovery procedure when MCU connection is lost
   - Implement auto-reconnect mechanism

3. **Improve time synchronization**:
   - Add reasonability check for received time
   - Use local timekeeping when NTP is unavailable

4. **Optimize memory usage**:
   - PROGMEM string encoding is already used, but buffer sizes could be reviewed

These changes should significantly enhance the reliability and stability of the thermostat, especially in critical situations such as MCU connection loss, time synchronization issues, or unexpected system states.
```

krzysiek2788, 2025-07-11:
As I can see this project is not developed anymore, so maybe you should make a fork and make the changes by your own?

gwaitsi, 2025-09-08:
did you fork an improved version? i want to change my devices, but the python components no longer install

labodj, 2026-10-27:
For anyone still looking for a maintained fork: I’ve resumed development of [WThermostatBeca-labo](https://github.com/labodj/WThermostatBeca-labo) and published [v1.25.0-labo.4](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.25.0-labo.4), with ready-to-flash ESP8266 binaries.

It includes UART framing/timeouts and MCU initialization retry improvements, memory and input-validation fixes, updated Home Assistant discovery, optional administrator authentication, and BSSID locking with recovery.

The release has 58 native regression tests, official Home Assistant schema validation and passing firmware builds. It addresses several areas discussed here, although it does not implement every proposal in this issue or add support for every thermostat MCU.

Thanks to fashberg and the earlier contributors for the foundation. Feedback from supported ESP8266 devices is welcome.

IFR-RFI, 2026-11-07:
@labodj thanks for picking this firmware up. Your README sends problem reports to the fork's issues, but they are disabled on your repository, so I could not post this there. I'm posting here instead, where you asked for feedback. You may want to enable them.

**This code seems to still have the same problem. I saw it on `v1.23.beta1-fas`; I have not tested your firmware.** So this is a pointer from reading the code, not a bug confirmed on your release.

**The problem, on `v1.23.beta1-fas`:** with the thermostat in `auto`, a new target temperature from Home Assistant makes the module report `mode: heat`, `schedulesMode: off` and the new target. But the display stays in auto with the old setpoint and the relay does not activate. 0 of 15 attempts worked (3 thermostats, 5 attempts each, physical model BHT-002-GxLW, set the same in the firmware, `No Hardware Hack`, `mcuId` `IAYz2WK1th0cMLmL1.0.0`). Sending `auto`, then `heat`, then the target worked 5 of 5. Measurements are in #239.

**On `v1.23.beta1-fas` the cause is visible in the trace log:** only the temperature frame (`02 02`) goes out and the mode frame (`04 04`) never does, although the module reports `heat`. Sending both frames back to back through `mcucommand`, from `auto`, worked 5 of 5 on one thermostat: display in manual at the new target and relay active within about 5 s.

**Why it may still apply to your firmware (from reading your code, not tested):** `onChangeTargetTemperature` calls `schedulesMode->setSuppressOnChange()` before setting it to `off` ([`WBecaDevice.h`, lines 2070-2075 at `v1.25.0-labo.5`](https://github.com/labodj/WThermostatBeca-labo/blob/785f1fd/WThermostat/WBecaDevice.h#L2070-L2075)). That function matches `b53a414`, and I found nothing about this in your release notes or commit messages, though I may have missed something.

**Untested patch.** The change below compiles against `b53a414` and behaves correctly in a host test of the module logic only, not on hardware. Removing the suppression alone would send the old target, because `schedulesModeToMcu` calls `updateTargetTemperature`, so the requested value has to be stored first. The back-to-back test above supports the approach, but the patched firmware itself has not been flashed on any device.

```diff
--- a/WThermostat/WBecaDevice.h
+++ b/WThermostat/WBecaDevice.h
@@ -1916,14 +1916,15 @@
                bool force=false;
                if (schedulesMode->equalsString(SCHEDULES_MODE_AUTO)){
                        network->log()->trace(PSTR("Got new targetTemperature: Switching from Schdule to Manual"));
-                       schedulesMode->setSuppressOnChange();
+                       // schedulesModeToMcu() calls updateTargetTemperature(): keep the requested value
+                       targetTemperatureManualMode = targetTemperature->getDouble();
                        schedulesMode->setString(SCHEDULES_MODE_OFF);
                        force=true;
                }
                if (schedulesMode->equalsString(SCHEDULES_MODE_OFF)){
                        // only set targetTemperatureManualMode and targetTemperatureManualModeToMcu() if current mode is Manual
                        if (force || !WProperty::isEqual(targetTemperatureManualMode, this->targetTemperature->getDouble(), 0.01)) {
-                               targetTemperatureManualMode = this->targetTemperature->getDouble();
+                               if (!force) targetTemperatureManualMode = this->targetTemperature->getDouble();
                                network->log()->trace((String(PSTR("onChangeTargetTemperature, temp: "))+String(targetTemperatureManualMode)).c_str());
                                targetTemperatureManualModeToMcu();
                                schedulesMode->setString(SCHEDULES_MODE_OFF);
```

**Quick check, if you have a thermostat at hand:** put it in `auto` with the room a few degrees below the new target, publish to `.../cmnd/things/thermostat/properties/targetTemperature`, and watch the display and relay. The trace log should show whether a `04 04` frame goes out. If it behaves correctly on your firmware, then my reading is wrong, and that would be useful to know too.

**Workaround I use on v1.23 beta:** a Home Assistant automation reacts to the `call_service` event for `climate.set_temperature` and, if the thermostat is in `auto`, resends `auto`, `heat` and the temperature, with pauses of 3 s and 4 s. It is tested with one command and with three in a row, and once from the real card in the HA interface.

labodj, 2026-11-08:
@IFR-RFI Thanks for the detailed report and for clearly separating your hardware tests from your review of our code.

I checked v1.25.0-labo.5 and reproduced the missing manual-mode command in host tests using the production property callbacks and UART serializers. Your diagnosis is correct: suppressing the schedulesMode callback changes the ESP’s internal state without sending the mode change to the thermostat MCU.

I also confirmed your point about the patch ordering. Removing the suppression alone restores the old manual setpoint; preserving the requested temperature before changing mode produces the expected manual-mode frame followed by the new temperature frame. This passed on both host and 32-bit test configurations. I have not verified the patched firmware on hardware yet.

You were also right about Issues being disabled on the fork. I’m enabling them and preparing the fix with a regression test for this exact sequence. The existing tests missed it.

One difference worth mentioning: the concatenated, already checksummed mcucommand payload from #239 is rejected by this fork’s stricter frame validation. That does not affect your diagnosis of the normal setpoint command.

Thanks again — this is an actionable report, and it helped identify a real defect.

labodj, 2026-11-08:
@IFR-RFI The fix is now available in [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo), and Issues are enabled on the fork. Thanks again for identifying the missing mode frame and the setpoint-ordering detail.

IFR-RFI, 2026-11-09:
@labodj thanks for the quick fix. I flashed `v1.26.0-labo` over OTA on one thermostat (BHT-002-GxLW, `mcuId` `IAYz2WK1th0cMLmL1.0.0`, `No Hardware Hack`, previously `v1.23.beta1-fas`) and it works:

- **From `auto`, new target 24 °C via Home Assistant:** the trace log shows `schedulesModeToMcu off` with the `04 04` frame, then `02 02` with 24 °C. The display went to manual at 24 °C, the relay closed and a separate device reading the relay output saw demand about 4 s later. I ran it twice, the second time with a person watching the display.
- **In manual, 24 → 25 → 26 °C:** only `02 02` goes out and the display follows.
- **Back to `auto`:** the display returns to the schedule (15 °C) and the relay opens.
- **`off`:** works.

The settings, MQTT topics, Home Assistant entities and the schedule were all kept. I'll leave it a few days before updating the other three.